osrf/mbari_wec_gz #17: Add hydrodynamics plugin to heave cone

chapulina, 2022-04-19:
We can use Gazebo's [Hydrodynamics](https://ignitionrobotics.org/api/gazebo/6.9/classignition_1_1gazebo_1_1systems_1_1Hydrodynamics.html) plugin on the heave cone, since it will be fully submerged. 

We'll need to either extend the existing hydrodynamics plugin, or to develop a new plugin for the free-surface hydrodynamics on the buoy itself. This is out of the scope of this issue.

quarkytale, 2022-08-18:
@andermi @hamilton8415 Is there any data regarding what's the velocity of the buoy when a known amount of force is applied? It'll help me set some values in the unit tests.

andermi, 2022-08-18:
> the velocity of the buoy when a known amount of force is applied

@hamilton8415 could we get something in the right ballpark by looking at the load cell and range_finder in the deployment logs?

hamilton8415, 2022-08-18:
Yes, we certainly have tons of data that should be useful for this.  In terms of "when a known amount of force is applied", the load-cell isn't the entire story as the waves and water and all of that are also forcing the buoy.  Although as an order of magnitude, it's probably about right (the buoy doesn't fly into the sky for instance...).  Another even simpler ballpark for vertical velocity ranges is the vertical speed of the sea-surface.  eta_dot = A*omega*cos(omega*t)...  Typical values would be A = 0.5-4 and  omega = 2*pi/T where T is the wave period between 4-16 seconds...  Let me know if any of this helps...

hamilton8415, 2022-08-18:
@quarkytale, andermi and I spoke about this a bit more and would like to know a bit more about what you need to proceed.  Might be best to have a phone/zoom call if you'd like...

quarkytale, 2022-08-19:
So now I have some basics tests in the upstream plugin. One insight I got from working on those were the added mass to be proportional to the rigid body mass. Initially I had a very small rigid body mass and the body flew away with nan velocities, which is similar to what I was facing with the heave cone. I think checking SI units throughout the project would help and remove any doubts there.

Otherwise, I had in mind to add certain checks here to validate real-world values. For example a range of forces which are acting on the buoy causing a range of velocities in z-direction(say). I think I can duplicate those forces in the plugin using the current vector or the directly applying it to any of the link in the buoy model.

This is definitely better discussed on call, I'll set up something!

mabelzhang, 2023-01-18:
Summary tasks from group discussion:

Looks like the existing hydrodynamics plugin can handle this.
- [ ] (@hamilton8415) Coefficients: Rob McEwen working on decent estimates of the needed coefficients.
- [ ] (@hamilton8415) PR: plugin activated in fs_hydrodynamics (#103), but may break into separate PR. Wait until #81, then create new branch for viscous drag.
- [ ] (@quarkytale) Add hydrodynamics plugin to SDF, may need one per link

closed 2023-01-31 by andermi