osrf/mbari_wec_gz #21: Free surface hydrodynamics

chapulina, 2022-04-19:
Gazebo's [Hydrodynamics](https://ignitionrobotics.org/api/gazebo/6.9/classignition_1_1gazebo_1_1systems_1_1Hydrodynamics.html) plugin is currently well-suited for subsurface links. We need to either extend that plugin, or create a new one that implements hydrodynamic forces at the surface, taking into account wave exciting, radiation and diffraction forces.

hamilton8415, 2022-05-31:
I have gotten a start on implementing some classes that generate incident waves and compute the resulting forces on the buoy as a result of those waves, and the forces that result from the buoy moving.  So far this is contained in two non-Gazebo classes, and is set up such that a different incident wave-generator class could be used reasonably easily.  The work of Rhys Mainwaring looks promising too.   I took a look at his code and it seems he's done a careful job of computing the buoyancy force based upon the instantaneous position of the buoy and water surface.  This is a fancy way of doing things, beyond what is usually done in ocean engineering.  On the other hand, for the damping part of the fluid-body interaction, it appears there is more reliance on heuristics.  In particular the damping that ordinary comes from the bodies tendency to generate waves is lumped in with the quadratic damping due to the viscosity of the fluid.  The FS_Hydrodyamics class implements a more physics based approach to generating the wave radiation forces, and could possibly be used in place of the Heuristics Rhys is using.  Those physics rely on coefficients generated by another program (WAMIT) that characterize the wave-body interaction dynamics.  The WAMIT users manual has some good background:  https://www.wamit.com/manualv7.4/wamit_v74manual.html    My intention at this point is to develop a simple Gazebo plugin on top of this code for demonstration purposes anyway, and we can see if it's a better path to refine that or try to collaborate with Rhys. He's done a lot of good work there, his wave-generation and visualization stuff is great, and he's included some of the non-linear fluid damping terms that are likely to be important.    Work is in branch hamilton8415/fs_hydrodynamics  At the moment I have CMakeLists.txt symbolically linked to a local CMakeLists file for ease of development on these non-Gazebo components...

mjcarroll, 2022-10-31:
@hamilton8415 to open PR off of his branch.

hamilton8415, 2023-01-17:
As threatened, I broke out the wave-body interaction force computation into a library independent of the Gazebo simulator.  This is now here:  https://github.com/hamilton8415/FreeSurfaceHydrodynamics

At this time in the hamilton8415/fs_hydrodynamics branch, the CI script installs this library from Debians in a PPA I made for this, so the building/installing is straightforward.  Seems to be working fine, we will see what issues of version-itis ensue...

Today I ended up having a zoom-call with Rhys Mainwaring (not a holiday in England, but it is raining...).  He is returning his focus to his wave-simulation, and by breaking these computations out he may be able to incorporate some of this work into his work. He's doing a heck of a job and there's still a good chance for this project to benefit from his plugins (and associated wave-rendering), he totally understands we needed a basic solution without his material though, so all is good.  

For now, I am keeping hamilton8415/fs_hydrodynamics up to date with the humble-garden branch, so will be ready to merge this into main as soon as humble-garden is merged.

closed 2023-03-23 by andermi