osrf/mbari_wec_gz #30: Aggregate data in ECM and use system to publish to ROS 2

chapulina, 2022-05-02:
As an alternative to publishing all values as doubles and bridging each to a ROS topic, the approach here would be to put the values in a custom component, and then have a single system that's ROS-enabled and publishes a message with all of them.

andermi, 2022-05-05:
@hamilton8415 to assist with any of these; I can make the first Controller as an example for the others. @rhenthorn 
 let me know if this looks good to you.
@chapulina should we open a separate issue for commands these controllers would need to handle?

We can break these into separate PR's that make sense

Todo:
- [ ] @andermi Add SCRecord data to ECM from PneumaticSpring
- [ ] @rhenthorn Add PCRecord, BCRecord data to ECM from ElectroHydraulicPTO
- [ ] @rhenthorn Add TFRecord data to ECM
- [ ] @andermi Add XBRecord data to ECM
- [ ] @andermi Publish SCRecord data from ECM in SpringController ROS2 System Plugin
- [ ] @rhenthorn Publish PCRecord data from ECM in PowerController ROS2 System Plugin
- [ ] @rhenthorn Publish BCRecord data from ECM in BatteryController ROS2 System Plugin
- [ ] @rhenthorn Publish TFRecord data from ECM in TrefoilController ROS2 System Plugin
- [ ] @andermi Publish XBRecord data from ECM in CrossbowController ROS2 System Plugin

andermi, 2022-06-08:
@hamilton8415 @rhenthorn Can you provide a description of the status fields for the __Records? (or point me to the docs for them)

hamilton8415, 2022-06-08:
My understanding is you're asking for the bit-field descriptions in the status words from the various converters.  It's a closely held secret!!  ;)  

Kidding of course, a good start is from the source code for each controller, copied in here and generally in "types.h" in each of the controller firmwares on bitbucket:

SC_
struct StatusBits {
    unsigned int ReliefValveRequest : 7;  //Request to open/close valve
    unsigned int ReliefValveStatus : 1;      // Status of Relief valve open/close
    unsigned int PumpRequest : 1;   // Request to turn pump on or off
    unsigned int PumpStatus : 1;  // Status of pump switch
    unsigned int PumpOverTemp : 1;  // Status of pump OverTemp signal
    unsigned int PumpToggle : 1;  // Status of pump Toggle.
    unsigned int TetherPowerRequest : 1;  //Request to turn tether power on or off
    unsigned int TetherPowerStatus : 1;  //Status of tether power relay
    unsigned int LR_Fault : 1;  //Status of LRF fault input
    unsigned int AUX_Fault : 1;  //Status of AUX fault input
};

typedef  union
{
    unsigned int status;
    struct StatusBits bits;
}  STATUS;

PC_
struct PowerConverterStatusBits {
    unsigned int Mode : 2;  //controller mode.  specified by defines in config.h (GENERATOR_MODE, TORQUE_MODE)
    unsigned int TorqueCmdLimited : 1;  //defines what is limited the Torque Command (0 = Not limited, 1 = Rate, 2 = Min, 3 = Max).
    unsigned int BattSwitchRequest : 1;  //Indicates state of battery switch that the user desires. 0 = off, 1 = on
    unsigned int BattSwitchSetting : 1;  //Indicates instantaneous setting of Battery Switch
    unsigned int SlowSwitchSetting : 1;  //Indicates instantaneous setting of Bettery Slow Switch
    unsigned int EXT_Voltage_Detect : 1; //Indicates if >170V is available on outside connector
    unsigned int GainScheduleMode : 1;      // 0 = off.  1 = on.
    unsigned int OverCurrentShutdown : 1;   //1 indicates drive is in overcurrent shutdown.
    unsigned int OverVoltageShutdown : 1;   //1 indicates drive is in overvoltage shutdown.
    unsigned int SpringRangeValid : 1;  //1 indicates the current SC_Range value is valid (received within approximately SC_RANGE_VALID_TIMEOUT milliseconds)
    unsigned int PermissiveMode : 1;   //1 indicates user has selected "permissive mode" which allows WindingCurrent commands to be set even if SpringController Range packets aren't arriving
};

BC_
struct StatusBits {
    unsigned int Balancing : 5;   // Indicates number of batteries that are shunting power
    unsigned int StopCharge : 5;  // Indicates number of batteries requesting stop-charge
    unsigned int PowerSupply : 1;  //Indicates state of 24V load switch. 0 = off, 1 = on
    unsigned int Bender : 2;      // Indicates state of Bender operation.  0 = off, 1 = on, 2 = auto (cycling)
    unsigned int BenderRequest : 2;   //Indicates the Bender State request, changed only by CAN rcvd interrupt except for initialization.
    unsigned int BenderSwitchStatus : 1;  //Indicates a fault with the bender power switch
};

TC_
enum STATES{ST_MAXON_POWERED_DOWN,
            ST_MAXON_POWERED_UP_AND_DISABLED,
            ST_MAXON_ENABLED,
            ST_MOVING_HOME_TOWARDS_CURRENT_POSITION,
            ST_VELOCITY_MOVE_TOWARDS_TARGET_POSITION,
            ST_MOVING_HOME_TOWARDS_TARGET_POSITION,
            ST_PERFORMING_BACKUP_MOVE,
            ST_DOORS_SUCCESSFULY_MOVED
};  //Bit mapping for "State" in StatusBits structure.

struct StatusBits {
    unsigned int ChargeMode : 2;  // Offset = 0
    unsigned int ChargeEnabled : 1;  // Offset = 2
    unsigned int MaxonPwrMode : 2;  // Offset = 3
    unsigned int MaxonPower : 1;  // Offset = 5
    unsigned int MaxonEnabled : 1;  // Offset = 6
    unsigned int DoorPositionTarget : 1;  // Offset = 7
    unsigned int DoorPositionActual : 1;  // Offset = 8
    unsigned int State : 4;  // Offset = 9
    unsigned int Maxon_CommError : 2;  // Offset = 13
};

closed 2022-07-08 by chapulina