osrf/mbari_wec_gz #31: Add physical properties to mbari_wec model

andermi, 2022-05-04:
I've put properties from Fran&ccedil;ois in a spreadsheet. These should be added to the mbari_wec model and/or accounted for in plugins.
[mbari_wec_properties.xlsx](https://github.com/osrf/buoy_sim/files/8626607/mbari_wec_properties.xlsx)

quarkytale, 2022-05-04:
Thanks for compiling and sharing this!
Making sure if I'm reading the values correctly, the added mass (closed/open) is in z-direction right? Also could you please help me identify the individual terms of Inertia (kg*m^2).

chapulina, 2022-05-05:
I'm going to work on the Inertia, weight and CoG properties for all links in a standalone PR now. It would help me with #26 so I can work with more realistic dynamics while updating the cable. And it should help your buoyancy efforts too, @quarkytale .

> could you please help me identify the individual terms of Inertia (kg*m^2). I'm assuming its the 3x3 rotational inertia matrix with the format:

I'm going with this same assumption and will verify visually if it looks reasonable.

andermi, 2022-05-05:
> the added mass (closed/open) is in z-direction

correct!

> 3x3 rotational inertia matrix with the format:

also correct!

chapulina, 2022-05-11:
Some values we're currently estimating but could need more precise numbers in the future:

- Piston moments of inertia
- Have cone inertia without Trefoil
- Trefoil inertia

chapulina, 2022-05-11:
For record-keeping, we've moved the CoB of the buoy a bit up to `2.46 m` instead of `2.16 m` in #33 so the box volume matches the shape of the buoy a bit better. We may want to tweak this in the future.

closed 2022-10-31 by mjcarroll