osrf/mbari_wec_gz #49: Add wave model

andermi, 2022-05-18:
We would like to have a physical and visual representation of waves and wave-body interaction. Some implementations can be discussed/tracked here until we choose the appropriate path forward.

1. FFT with OpenGL 3.3
- [Part One](https://www.keithlantz.net/2011/10/ocean-simulation-part-one-using-the-discrete-fourier-transform/)
- [Part Two](https://www.keithlantz.net/2011/11/ocean-simulation-part-two-using-the-fast-fourier-transform/)
- [GitHub](https://github.com/Postrediori/OceanSimulation)

andermi, 2022-05-18:
For 1.
Pros:
- Visually appealing
- Very efficient
- Has access to deformable mesh vertices
Cons:
- Doesn't seem to fit with existing Gazebo GLSL program interface like in 2.

srmainwaring, 2022-09-13:
Hi @andermi, I've started to look into how you might combine the wave rendering and hydrostatics model from [asv_wave_sim](https://github.com/srmainwaring/asv_wave_sim) with the proposed free surface wave-body interaction model being developed in buoy_sim. There's still a lot to resolve but I've made a start getting the environment set up with a couple of test examples and worlds in this branch: 

-https://github.com/srmainwaring/buoy_sim/tree/srmainwaring/gz-sim7-fs_hydro

There are two example wave models available. One is the randomised wave model based on Tessendorf's work, the other is a monochromatic plane wave with period 8s and amplitude 2m. I have also added a test model based on the WEC-Sim ellipsoid used to demonstrate their non-linear hydrostatics approach. This is for eventual benchmarking against their simulation.

I'll add some notes to the branch describing how to get the system set up.

Figure: [mbari_wec_waves.sdf](https://github.com/srmainwaring/buoy_sim/blob/srmainwaring/gz-sim7-fs_hydro-waves/buoy_gazebo/worlds/mbari_wec_waves.sdf) configured to use the [monochromatic_waves]( https://github.com/srmainwaring/buoy_sim/blob/srmainwaring/gz-sim7-fs_hydro-waves/buoy_gazebo/models/monochromatic_waves) model.

[image]

closed 2022-10-31 by mjcarroll