osrf/mbari_wec_gz #66: GUI dashboard for the buoy

chapulina, 2022-06-24:
It would be handy to have a GUI dashboard that allows sending commands to the buoy and reading various parameters.

For example, for the command added on https://github.com/osrf/buoy_sim/pull/51#issuecomment-1144213629, we'd have a spin box where the user can input a duration and the GUI sends a ROS service request.

Here's a very basic GUI plugin to get started with: https://github.com/gazebosim/gz-gui/tree/ign-gui6/examples/plugin/hello_plugin. The plugin here would also need to link to `rclcpp`.

andermi, 2022-06-27:
can also make use of `#include <buoy_msgs/interface.hpp>`

andermi, 2022-08-12:
we need this just so we can call it `buoy_gui`

quarkytale, 2022-08-23:
Can also make use of `pbcmd` [buoy_msgs #24](https://github.com/osrf/buoy_msgs/pull/24)

closed 2022-10-31 by mjcarroll