osrf/mbari_wec_gz #84: Use REP103 SI Units/Convention

andermi, 2022-08-18:
Need to make a pass over the whole system (internal calculations, state data, published messages) to make sure we are consistent with ROS2 standard SI units and conventions: [REP103](https://www.ros.org/reps/rep-0103.html)

- [ ] PolytropicPneumaticSpring
- [ ] ElectroHydraulicPTO
- [ ] XbowAHRS
- [ ] PowerController
- [ ] SpringController

hamilton8415, 2022-08-18:
Thanks Micheal.  As further background Michael, Henthorn, and myself spoke about this some yesterday and our issue is that some of the existing buoy units are Imperial units (which we're still very comfortable with! :).  These units turn up in logged data and user commands issueable from the command line of the buoy computer, items we don't want to change...  So the path that looks best is to have the simulator and ROS2 messages use SI units common to ROS/Gazebo world, and then simulate our command line and logging tools as needed and do the conversion to Imperial/SI units there.  Ultimately the goal is to have the simulator and buoy environment appear indistinguishable, so unless we change the Imperial units on the buoy code, this seems to be the best place to make the conversion.  It does mean we need to maintain the simulated logging and command features to match changes subsequently made on the buoy (which should be limited), but there are structural reasons that changing the legacy buoy code is problematic so we think this is the best path.

TL;DR:  What Michael said, all SI in gazebo and ROS2 messages and we'll translate in the simulation of the legacy buoy user interface.  I looked and it's not that bad a list since all of our electrical items are already SI...

mabelzhang, 2023-01-12:
Ticket on hold.
Notes from group discussion:
Will "keep pressure and angular rate in the ElectroHydraulicPTO plugin in psi and RPM respectively.  Seems harmless enough and is easier to conflate with our experience with the real system during test/debug." The RPM change will be invasive.