osrf/mbari_wec_gz #95: How to workaroudn the issues about "Unable to find uri[package://buoy_description/models/mbari_wec]"

KunYi, 2022-09-29:
I follow [buoy_entrypoint](https://github.com/osrf/buoy_entrypoint) steps to run the buoy_sim.
but got the below error, 
have anyone give some tips for how to workaround the issues?

my host environments
```
Ubuntu: 22.04/AMD64
declare -x GZ_VERSION="fortress"
declare -x ROS_DISTRO="humble"
declare -x ROS_LOCALHOST_ONLY="0"
declare -x ROS_PYTHON_VERSION="3"
declare -x ROS_VERSION="2"
declare -x IGNITION_VERSION="fortress"
declare -x IGN_GAZEBO_RESOURCE_PATH="/home/kunyi/buoy_ws/install/buoy_gazebo/share/buoy_gazebo/worlds:/home/kunyi/buoy_ws/install/ros_gz_sim_demos/share:/home/kunyi/buoy_ws/install/buoy_description/share/buoy_description/models:/home/kunyi/buoy_ws/install/buoy_description/share"
declare -x IGN_GAZEBO_SERVER_CONFIG_PATH="/home/kunyi/buoy_ws/install/buoy_gazebo/share/buoy_gazebo/gazebo/server.config"
declare -x IGN_GAZEBO_SYSTEM_PLUGIN_PATH="/home/kunyi/buoy_ws/install/buoy_gazebo/lib"
```
Error Messages.
```
[robot_state_publisher-3] Error [SDF.cc:160] Tried to use callback in sdf::findFile(), but the callback is empty.  Did you call sdf::setFindCallback()?
[robot_state_publisher-3] [ERROR] [1664422085.714176737] [sdformat_urdf]: Unable to find uri[package://buoy_description/models/mbari_wec]
[robot_state_publisher-3] [ERROR] [1664422085.715995362] [sdformat_urdf]: A model must have at least one link.
[robot_state_publisher-3] [ERROR] [1664422085.716005767] [sdformat_urdf]: A model must have at least one link.
[robot_state_publisher-3] [ERROR] [1664422085.716008607] [sdformat_urdf]: FrameAttachedToGraph error: scope context name[] does not match __model__ or world.
[robot_state_publisher-3] Failed to parse robot description using: sdformat_urdf_plugin/SDFormatURDFParser
[robot_state_publisher-3] terminate called after throwing an instance of 'std::runtime_error'
[robot_state_publisher-3]   what():  Unable to initialize urdf::model from robot description

```
[image]

quarkytale, 2022-09-29:
Hi there, did you source the workspace after building it? If yes, then can you send the output of `echo $IGN_GAZEBO_RESOURCE_PATH`.

KunYi, 2022-09-29:
> 
Hi @quarkytale , thank you for your response
see the picture
[image]

quarkytale, 2022-09-29:
Thanks for sharing. The paths seems to be set correct, Gazebo should be stripping the "package://" and taking care of it. One workaround is to set `export SDF_PATH=$IGN_GAZEBO_RESOURCE_PATH` and try launching again. In the meantime, I'll try reproducing the issue.

KunYi, 2022-09-29:
@quarkytale, 
 thanks  for your answer!
 now going next step. 
 it will show ignition gazebo for select world
[image]

```
kunyi@kunyi-ThinkPad-P53:~/buoy_ws$ export SDF_PATH=$IGN_GAZEBO_RESOURCE_PATH
kunyi@kunyi-ThinkPad-P53:~/buoy_ws$ ros2 launch buoy_gazebo mbari_wec.launch.py
[INFO] [launch]: All log files can be found below /home/kunyi/.ros/log/2022-09-29-12-08-00-970191-kunyi-ThinkPad-P53-20783
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [20784]
[INFO] [parameter_bridge-2]: process started with pid [20786]
[INFO] [robot_state_publisher-3]: process started with pid [20789]
[parameter_bridge-2] [ros_ign_bridge] is deprecated! Redirecting to use [ros_gz_bridge] instead!
[parameter_bridge-2] 
[parameter_bridge-2] [INFO] [1664424481.158519809] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[parameter_bridge-2] [INFO] [1664424481.160364013] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/world_demo/model/MBARI_WEC_ROS/joint_state (ignition.msgs.Model) -> /world/world_demo/model/MBARI_WEC_ROS/joint_state (sensor_msgs/msg/JointState)] (Lazy 0)
[parameter_bridge-2] [INFO] [1664424481.162011165] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/MBARI_WEC_ROS/pose (ignition.msgs.Pose_V) -> /model/MBARI_WEC_ROS/pose (tf2_msgs/msg/TFMessage)] (Lazy 0)
[parameter_bridge-2] [INFO] [1664424481.163331742] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/MBARI_WEC_ROS/pose_static (ignition.msgs.Pose_V) -> /model/MBARI_WEC_ROS/pose_static (tf2_msgs/msg/TFMessage)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1664424481.338746002] [sdformat_urdf]: SDFormat link [Buoy] has a <sensor>, but URDF does not support this
[robot_state_publisher-3] The root link Buoy has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] Link PTO had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'Universal' into a fixed joint
[robot_state_publisher-3] Link Piston had 1 children
[robot_state_publisher-3] Link tether_top_0 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_0' into a fixed joint
[robot_state_publisher-3] Link tether_top_1 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_1' into a fixed joint
[robot_state_publisher-3] Link tether_top_2 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_2' into a fixed joint
[robot_state_publisher-3] Link tether_top_3 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_3' into a fixed joint
[robot_state_publisher-3] Link tether_top_4 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_4' into a fixed joint
[robot_state_publisher-3] Link tether_top_5 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_5' into a fixed joint
[robot_state_publisher-3] Link tether_top_6 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_6' into a fixed joint
[robot_state_publisher-3] Link tether_top_7 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_7' into a fixed joint
[robot_state_publisher-3] Link tether_top_8 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_8' into a fixed joint
[robot_state_publisher-3] Link tether_top_9 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_9' into a fixed joint
[robot_state_publisher-3] Link tether_top_10 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_10' into a fixed joint
[robot_state_publisher-3] Link tether_top_11 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_11' into a fixed joint
[robot_state_publisher-3] Link tether_top_12 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_12' into a fixed joint
[robot_state_publisher-3] Link tether_top_13 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_13' into a fixed joint
[robot_state_publisher-3] Link tether_top_14 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_top_joint_14' into a fixed joint
[robot_state_publisher-3] Link tether_bottom_0 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_bottom_joint_0' into a fixed joint
[robot_state_publisher-3] Link tether_bottom_1 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_bottom_joint_1' into a fixed joint
[robot_state_publisher-3] Link tether_bottom_2 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_bottom_joint_2' into a fixed joint
[robot_state_publisher-3] Link tether_bottom_3 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_bottom_joint_3' into a fixed joint
[robot_state_publisher-3] Link tether_bottom_4 had 1 children
[robot_state_publisher-3] Converting unknown joint type of joint 'tether_bottom_joint_4' into a fixed joint
[... cut ...]
[robot_state_publisher-3] [INFO] [1664424481.360921115] [robot_state_publisher]: Floating joint. Not adding segment from tether_bottom_3 to tether_bottom_4.
[robot_state_publisher-3] [INFO] [1664424481.360926394] [robot_state_publisher]: Floating joint. Not adding segment from tether_bottom_4 to HeaveCone.
[ign gazebo-1] Xlib:  extension "NV-GLX" missing on display ":1".
[ign gazebo-1] libEGL warning: DRI2: failed to create dri screen
[ign gazebo-1] libEGL warning: DRI2: failed to create dri screen

```

closed 2022-09-30 by KunYi