osrf/mbari_wec_gz #113: Parameterization at startup

mabelzhang, 2023-01-12:
Offer a simple and obvious way to allow the user to specify a number of parameters at start-up.

Things to parameterize:
- [ ] Wave field - wave height, wave period
- [ ] Scale factor
- [ ] Initial battery voltage (state of charge)
- [ ] Make heave cone doors fixed joint in SDF, and set open/closed state at startup via parameter. The open/closed state affects a lot of parameter changes in the SDF.
- [ ] Enable batch runs, even if with simple bash scripting

Implementation options:
- SDF (currently generated at build time)
- Launch file parameter (if just a few)
- YAML file, which can handle arrays

andermi, 2023-01-19:
@hamilton8415 is the scale factor something to set for an entire run or more like the gain scheduler sending a scale factor to refresh the command timeout? Or even exactly like the gain scheduler sending various scale factors over a period of time?

hamilton8415, 2023-01-19:
It works a few ways, the scale factor is a multiplier to the built in damping coefficient or the power-take off device.  In most simulation cases I think the usual pattern would be to set a scale factor at start-up, and collect the results for a 10-15 minute sample period.  It is possible to change the scale-factor during runs hrough the pbcmd "scale" command.  Same command on the buoy and also the simulator....  Finally, we do have an application we often run on the buoy that is changing the scale factor every 30 minutes or so based on measurements from a local wave-height measurement buoy.  @andermi can comment further, but I believe that is ros application that uses the ros services, and that michael actually used an early version of this simulator before it was deployed on the buoy earlier this summer.

closed 2023-03-06 by andermi