osrf/mbari_wec_gz #114: Log files

mabelzhang, 2023-01-12:
Replicate csv log files from physical buoy system.

On the physical system, filenames and folders are specific to the time of day. Think about the best way to do this for a faster-than-real-time simulated system.
hamilton8415: "Perhaps it's as simple as each simulation start creates a new logfile directory with a name specific to the start-time, then the filenames proceed forward from there representing simulation time."

Assign to rhenthorn upon agreement

rhenthorn, 2023-01-23:
As long as the data messages contain simulation time there should not be an issue.

rhenthorn, 2023-02-27:
I have begun working on this issue. Created henthorn/sim_pblog branch in buoy_msgs repo.

Used the python controller template as a starting point for this node - handy as the template controller subscribes to all the topics needed.

andermi, 2023-02-28:
@rhenthorn would you please add a launch argument (to controller.launch.py or whatever you may have changed the name to) to specify a custom log directory?

rhenthorn, 2023-03-16:
The node sim_pblog has been added to buoy_msgs repo branch 'henthorn/sim_pblog'. Some items to resolve:

1. Add a custom log folder as a launch argument
2. First field 'epoch seconds' in the log file (use simulation time - starts at 0.0, or add sim time to current epoch seconds?)
3. Integrate into buoy_gazebo to launch sim_pblog with the simulation

Thanks everyone for your help and patience.

rhenthorn, 2023-03-16:
Also add to items to resolve

1. insert dependencies in packages.xml

andermi, 2023-03-17:
If you wouldn't mind starting a PR in the buoy_msgs repo, we could start reviewing :)

mabelzhang, 2023-03-27:
PR https://github.com/osrf/buoy_msgs/pull/44

closed 2023-05-29 by andermi