osrf/mbari_wec_gz #121: Create controllers to simulate heave cone and battery messages

mabelzhang, 2023-01-23:
Create messages for heave cone (TFRecord) and battery (BCRecord) in [buoy_interfaces/msg](https://github.com/osrf/buoy_msgs/tree/main/buoy_interfaces/msg)

Possibly add messages that match buoy, even if those values don't appreciably change, e.g. hydrogen in battery.

Assign one of @rhenthorn + @quarkeytale

mabelzhang, 2023-01-30:
Consider using SCRecord (spring controller) or PCRecord (power controller) as reference

andermi, 2023-02-01:
the SpringController is a good example. Just ignore the services for commands

some steps:
make a component .hpp file
- [ ] make a state struct like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/PolytropicPneumaticSpring/SpringState.hpp#L43)
- [ ] register it as a component like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/PolytropicPneumaticSpring/SpringState.hpp#L69)
in the plugin where the data originates
- [ ] access the state data component from the ECM like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/PolytropicPneumaticSpring/PolytropicPneumaticSpring.cpp#L525-L534) in preupdate
- [ ] set struct data like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/PolytropicPneumaticSpring/PolytropicPneumaticSpring.cpp#L591-L592) in preupdate
- [ ] write data to the ECM like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/PolytropicPneumaticSpring/PolytropicPneumaticSpring.cpp#L597) in preupdate
in the ROS2 controller plugin
- [ ] access state data component from ECM like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/controllers/SpringController/SpringController.cpp#L615-L625) in postupdate
- [ ] set ROS2 message data like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/controllers/SpringController/SpringController.cpp#L633-L639) from the state data struct in postupdate
- [ ] make a private struct to hold ROS2 stuff like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/controllers/SpringController/SpringController.cpp#L47-L58)
- [ ] make a struct for other private stuff like [this](https://github.com/osrf/buoy_sim/blob/bc7e9149afb9c8d7dfe259db8133c78742c1d278/buoy_gazebo/src/controllers/SpringController/SpringController.cpp#L79) but ignore stuff about services and commands. You probably still need the mutexes for low/high priority data access

andermi, 2023-02-01:
@rhenthorn @quarkytale let me know if you have any questions about the steps above

andermi, 2023-02-02:
the Trefoil Controller will probably need to borrow from some of the stuff in the [XBowAHRS](https://github.com/osrf/buoy_sim/tree/main/buoy_gazebo/src/controllers/XBowAHRS) controller to subscribe to IMU. Also, the Heave Cone link will need an IMU added to it (the IMU system plugin is already enabled)

andermi, 2023-02-10:
reopening for the TFController

closed 2023-02-18 by quarkytale