osrf/mbari_wec_gz #122: Buoyancy plugin update parameters

mabelzhang, 2023-01-23:
Current buoyancy plugin:
https://github.com/osrf/buoy_sim/blob/7723d2763deb9d07c2eace8c789c34c9b12ddc64/buoy_gazebo/gazebo/server.config#L19-L30

Use uniform buoyancy, not graded.
Set correct density.
Set whether to enable for specific links - need for heave cone + PTO only.

Not urgent.

mabelzhang, 2023-01-30:
Need to confirm volume, CoM, etc against latest documentation

andermi, 2023-02-17:
This is becoming more urgent I think.

Also, need to learn the way to turn off buoyancy plugin for the buoy float since that specific link is being handled in the WaveBodyInteractions plugin.

quarkytale, 2023-02-17:
Already started working on it. I believe there is an `<enable>` in the plugin for a model's link as well, but yet to try it out.