osrf/mbari_wec_gz #142: CI failing to detect 'ros-humble-actuator-msgs'

quarkytale, 2023-05-02:
All PRs on this repo are stuck on this error, even seeing the same while building the docker image on `mbari_wec`.

```
  apt: command [apt-get install -y ros-humble-actuator-msgs] failed
  apt: Failed to detect successful installation of [ros-humble-actuator-msgs]
Error: Process completed with exit code 1.
```

quarkytale, 2023-05-02:
> I guess it needs a port to humble? (edited)

Seems like its already added, https://github.com/ros/rosdistro/pull/36795

andermi, 2023-05-02:
well, somehow it didn't make it into apt after that

andermi, 2023-05-02:
at the very least, we aren't using anything related to the actuator_msgs stuff so I'm fine reverting and freezing ros_gz for now. We can just tie that commit to our release

andermi, 2023-05-02:
something like this: https://github.com/osrf/mbari_wec/pull/51

quarkytale, 2023-05-02:
Adding ros-testing sources list fixes our build, this is a temporary fix till a Humble sync is out (by this week).

andermi, 2023-05-02:
Please see my comment in #143

quarkytale, 2023-05-11:
Tested this after Humble sync, apt install worked and could build the latest `ros2` branch of `ros_gz` too!

closed 2023-05-11 by quarkytale