osrf/mbari_wec_gz #166: Temporarily add on_exit_shutdown to ros_gz humble branch until backport

andermi, 2023-09-20:
The `ros2` branch on [ros_gz](https://github.com/gazebosim/ros_gz.git) has moved passed us so we need to be on the `humble` branch. I just noticed that this [PR](https://github.com/gazebosim/ros_gz/pull/355) hasn't been backported to `humble` and notified @mjcarroll who is working on it now. The issue is that the batch script won't run more than one sim because it doesn't shutdown the current launch when `gz sim` exists.

In the meantime (until the backport):
in the `ros_gz` clone:
```
$ git checkout ros2
$ git pull
$ git checkout humble
$ git pull
$ git cherry-pick 63e85a952668b824d266740606a245c2a1db4d0c
```

and then `colcon build` the workspace again

mjcarroll, 2023-09-21:
Backports are in, I'll try to get a release out ASAP

andermi, 2023-09-21:
thanks a ton for the quick turnaround!

closed 2023-09-23 by andermi